SMotlaq/LoRa #4: Hello!Can it be implemented in FreeRTOS？

ssg0412, 2021-06-11:


SMotlaq, 2021-06-11:
I'm so sorry because I don't know :((
I haven't worked with FreeRTOS yet.

edilandrade, 2021-07-04:
Hello, thank you very much for sharing this lib, I implemented it and found some problem in receiving packets, when decreasing the delay in less than 300 ms starts to lose information to be received, would you have any solution?

SMotlaq, 2021-07-04:
> 
> 
> Hello, thank you very much for sharing this lib, I implemented it and found some problem in receiving packets, when decreasing the delay in less than 300 ms starts to lose information to be received, would you have any solution?

How do you check the packets? Do you use Keil debugger? or what?

edilandrade, 2021-07-09:
Estou usando o STMCubeIDE.

edilandrade, 2021-07-09:
Na imagem mostro a recepção dos pacotes em uma leitura de 800ms ainda perde alguns pacotes. 

[image]

edilandrade, 2021-07-09:
a cada intervalo de tempo verifico a recepção 
caso deixe no loop diretamente o mesmo não funciona.

[image]

SMotlaq, 2021-07-09:
Dude ENGLISH please :))

edilandrade, 2021-07-09:
I am using STMCubeIDE.

edilandrade, 2021-07-09:
at each time interval, reception is verified.
if you leave it in the loop directly, it doesn't work.

[image]

edilandrade, 2021-07-09:
In the image I show the reception of packets at a reading of 800ms still lose some packets.

[image]

edilandrade, 2021-07-09:
at every interval of time I check the reception
if you leave it in the loop directly, it doesn't work.

[image]

SMotlaq, 2021-07-09:
How fast is the transmitter?

edilandrade, 2021-07-11:
It's transmitting at 3000ms

[image]

SMotlaq, 2021-07-11:
Use DIO0 pin for receiver side. In EXTI callback use the LoRa_receive() and test it again.

edilandrade, 2021-07-17:
thank you very much, it worked using callback working perfectly.

SMotlaq, 2021-07-18:
Nice :)
Good luck ✌🏻✌🏻

closed 2021-07-18 by SMotlaq