SMotlaq/LoRa #5: Transmitting right after reading data

ImBetterThanNinja, 2021-07-05:
Hi, I wanted to respond to the data I just received.
I'm using external interrupt from DIO0 and there I'm receiving the data but when I'm trying to transmit data right after that it somehow breaks. It doesn't even go into exti callback anymore. Is it somthing you tried yet?
Here is my callback

`void HAL_GPIO_EXTI_Callback(uint16_t GPIO_Pin)
{
	HAL_GPIO_WritePin(GPIOC,GPIO_PIN_13,GPIO_PIN_RESET);
	if(GPIO_Pin == GPIO_PIN_14)
	{
		LoRa_receive(&myLoRa, read_data, 128);
		if(strstr(read_data,myIDcode))
		{
			sprintf(send_data, "received packet no.: %d",read_data[2]);
			//LoRa_transmit(&myLoRa, send_data, 128, 500);		
		}
	}
	HAL_GPIO_WritePin(GPIOC,GPIO_PIN_13,GPIO_PIN_SET);
}`

SMotlaq, 2021-07-05:
Hi, yes you are right. Because of a HAL_Delay(1) in the LoRa_transmit(...) function it doesn't work in interrupt callbacks.
Generally, you have to raise a flag in the callbacks and check it at the main loop to do something like sending your data. This is the professional way.
Another solution is to use a timer-based delay function instead of HAL_Delay.
And the last way is deleting the timeout detector in the LoRa_transmit(...). But it is not recommended 🙃

ImBetterThanNinja, 2021-07-06:
Thanks for your reply!
I still didn't manage to make two way communication work but at least that one issue is sorted out.
Now I'm having another problem with this bit of code:
```
while (1)
{
	sprintf(send_data, "%s%d",myIDcode,counter++);
	LoRa_transmit(&myLoRa, send_data, 128, 500);
	HAL_Delay(500);
}
```
Sometimes it works just fine but more often than not it sends only every second message. It doesn't crash or anything but LoRa_transmit() function takes less time to execute (on messages that module doesn't send out). Also when I tried to use LoRa_getRSSI(), on properly send messages it returns some reasonable value but on every second message it returns -164 (so it reads 0 from module). Am I missing something?

SMotlaq, 2021-07-06:
It looks fine and should work. Can you send me a video of your tries in Telegram?

ImBetterThanNinja, 2021-07-06:
Im not on telegram, sorry. I'll try it again and change antennas to proper 433MHz to make sure it's not the problem. It might be a mistake on my end because it works only for few minutes and then stops receiving entirely. If that doestn't work I'll try to record something, post it on YouTube and post link here.
Did you manage to get two way communication working with this library?

SMotlaq, 2021-07-06:
Yes I did and that worked for about 2 hours.
You can send me your entire project by e-mail: pilot.motlaq@gmail.com

closed 2021-07-06 by SMotlaq