SMotlaq/LoRa #8: Testing RA-02 SX1278 module with STM32L031

krupis, 2023-08-01:
I am trying to get this example project working on the STM32L031 development board:
https://www.st.com/en/evaluation-tools/nucleo-l031k6.html

I cannot fully understand how am I intended to test this. I have 2 identical STM32L031 devices and connected each STM32L031 to the RA-02 SX1278 lora module.

One is running LoRa transmit code:
```
  while (1)
  {
	  char*  send_data;
	  send_data = "Hello!";
	  if(LoRa_transmit(&myLoRa, (uint8_t*)send_data, 6, 100) == 1){
		  printf("Transmit ok \n");
	  }
		else{
			printf("Transmit failed \n");
		}
		HAL_Delay(1500);
    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }
```

The other is running receive code:
```
  LoRa_startReceiving(&myLoRa);
  uint8_t received_data[10];
  uint8_t packet_size = 0;
  while(1){
    packet_size = LoRa_receive(&myLoRa, received_data, 10);
    if(packet_size > 0 ){
    	printf("received data = %s \n",received_data);
    }
    HAL_Delay(500);
  }
```

The logs side by side:
[image]

SMotlaq, 2023-08-01:
Hello
Using `char*` without memory allocation is not recommended.
Please use `char*` as array and use `sprintf` and `snprintf` functions like this:
```C
char   send_data[20];
snprintf(send_data, 6, "Hello!");
LoRa_transmit(&myLoRa, (uint8_t*)send_data, 6, 100);
```
I think this should solve your problem.

krupis, 2023-08-02:
@SMotlaq I have updated the transmitter code as you have suggested but im afraid not much changed. I believe issue lies in the receiver code. Something is not right.

The receiver is constantly printing garbage even when I dont have transmitter powered.

[image]

SMotlaq, 2023-08-02:
Please send me your project and codes
pilot.motlaq@gmail.com

krupis, 2023-08-02:
@SMotlaq 
I can share my project through here. You access it via my github:
https://github.com/krupis/Lora-STM32

Additionally, I have put a scope on the DIO0 pin on the receiver device. It is never triggered but on the console it prints the received garbage data. Does your receiver code utilize DIO0 rx interrupt?

SMotlaq, 2023-08-02:
Have you checked what value LoRa_init() function returns?

krupis, 2023-08-02:
> Have you checked what value LoRa_init() function returns?

Yes. I have added the following to my code:
```
  LoRa_reset(&myLoRa);
  uint16_t lora_ret = LoRa_init(&myLoRa);
  printf("Lora return value = %u \n",lora_ret);
```

and the result:
```
Lora return value = 200 
Starting RX mode 
```

krupis, 2023-08-02:
If it would be helpful, I can connect logic analyzer to SPI pins and check what happens

SMotlaq, 2023-08-02:
1- Please reset the buffer first, using `memset` to check that we get any packet or not:
```C
uint8_t received_data[10];
memset(received_data,  NULL, 10);
```

2- Also please send a variable data from sender. For example "Hello 1", "Hello 2", ...

3- Print the exact values of the `received_data`. Not the character or ASCII.

krupis, 2023-08-03:
@SMotlaq 
I have modified my receiver code as following:

```
	if(MODA_LORA == 0){
		printf("Starting RX mode \n");
		LoRa_startReceiving(&myLoRa);
		uint8_t received_data[10];
		memset(received_data,  NULL, 10);
		uint8_t packet_size = 0;
		while(1){
		packet_size = LoRa_receive(&myLoRa, received_data, 10);
		if(packet_size > 0 ){
			for(int i = 0; i<packet_size;i++){
				printf("received data[i] = %u \n",i,received_data[i]);

			}
		}
		memset(received_data,  NULL, 10); // Clear the received data buffer after it is received
		packet_size = 0; // set packet size to 0 after receiving data.
		HAL_Delay(500);
		}
	}
```

and my transmitter code as following:
```
	else if(MODA_LORA == 1){
		printf("Starting TX mode \n");
		while (1)
		{
			char   send_data[20];
			for (int i = 0; i < 100; i++)
			{
				snprintf(send_data, 20, "test%d", i); // puts string into buffer
				printf("String to send = %s and length =%u \n", send_data,strlen(send_data)); // outputs so you can see it
				if(LoRa_transmit(&myLoRa, (uint8_t*)send_data, 6, 100) == 1){
					printf("Transmitting %s successful \n",send_data);
				}
				else{
					printf("Transmit failed \n");
				}
				HAL_Delay(2000);
			}

		/* USER CODE END WHILE */
		/* USER CODE BEGIN 3 */
		}
	}
```

The logs side by side:
[image]

As you can see lora receiver receives strange data. This data is being does not have anything to do with my lora transmitter since it will keep receiving:
```
received data[i] = 0 
received data[i] = 1 
received data[i] = 2 
received data[i] = 3 
received data[i] = 4 
received data[i] = 5 
received data[i] = 6 
received data[i] = 7 
received data[i] = 8 
received data[i] = 9 
```

Even when my tranmsitter is powered OFF.

SMotlaq, 2023-08-06:
I will run your code on my hardware as soon as possible and tell you the result.

krupis, 2023-08-08:
> I will run your code on my hardware as soon as possible and tell you the result.

Sounds good. Please let me know if you are able to reproduce the issue at all. :)

krupis, 2023-08-11:
@SMotlaq 
I have recently watched your Youtube video regarding this library.
I have noticed a few comments:
[image]

Perhaps my issue is related to this?

In my case, NVIC settings are as following
[image]

krupis, 2023-08-11:
@SMotlaq 
Additionally, I do not fully understand whether DIO0 signal is set high or low when message is received? Could you confirm whether it is active LOW or active HIGH? I havent been able to find this information

SMotlaq, 2023-08-11:
When the module receives a valid LoRa packet, it changes the DIO0 from LOW to HIGH (I don't know how long this pin stays HIGH). You can enable an EXTI in the rising edge mode to detect arriving packets.

You didn't use interrupts in your code, so I don't think that is the reason of your problem. But you can test it. The polling mode is not recommended. It's better to use interrupts.

krupis, 2023-08-12:
> When the module receives a valid LoRa packet, it changes the DIO0 from LOW to HIGH (I don't know how long this pin stays HIGH). You can enable an EXTI in the rising edge mode to detect arriving [...]

I do not fully understand what you mean. In your ```LoRa_init``` function, you have the following:
```
		// DIO mapping:   --> DIO: RxDone
			read = LoRa_read(_LoRa, RegDioMapping1);
			data = read | 0x3F;
			LoRa_write(_LoRa, RegDioMapping1, data);
```

So from what I understand, you enable the RX interrupt.

As I have previously mentioned, when I probe the DIO0 pin, it is never toggled HIGH so that should mean that no data is being received, but some garbage is being printed. Can you help me understand how this works?

Would you be able to provide any clues on how to further debug the issues with my lora receiver? Why would it be constantly receiving:
```
received data[i] = 0 
received data[i] = 1 
received data[i] = 2 
received data[i] = 3 
received data[i] = 4 
received data[i] = 5 
received data[i] = 6 
received data[i] = 7 
received data[i] = 8 
received data[i] = 9 
```
and no DIO0 interrupt received?

At this point I dont even know what should I test next.

closed 2024-07-14 by SMotlaq