SMotlaq/LoRa #10: Failed to receive LoRa packets with SF 12

IoTThinks, 2024-01-22:
Hi all,
I have tested to send packet from Sandeep's Arduino LoRa (ESP32) to SMotlag's LoRa (STM32).
When SF = 7, STM32 can receive packets from ESP32.
However, when SF=12, then STM32 does not receive packets from ESP32.

Compare the set Spreading Factor function, I guess this library misses the part to set LDO (Low data optimization).
This library
https://github.com/SMotlaq/LoRa/blob/master/LoRa/LoRa.c#L184

Sandeep's library
https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L573

RadioLib's library
https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1278.cpp#L188

Have you tested the receiving with SF12?
Thanks a lot.

IoTThinks, 2024-01-22:
I have attempted to creat a fix. But not successful yet.

Let me test your patch tomorrow morning.

IoTThinks, 2024-01-23:
I tested just now.
The fix works for SF7-10 and SF12.
SF11 seems not working yet. Still show wrong data for SF11.

[image]

SMotlaq, 2024-01-23:
Thanks. Would you mind sending your code please?

IoTThinks, 2024-01-24:
My code is not working yet.

Let me debug and try to fix based on your fix yesterday.

Hope to complete by this Friday.

IoTThinks, 2024-01-25:
I think there is still missing code here and there comparing to RadioLib and ArduinoLoRa.
However, this is a dirty trick.
SF7-11 should not have LDO on. SF12 should have LDO on.
((1 << _LoRa->spredingFactor) / ((double)BW[_LoRa->bandWidth] will return 16.3... so it will larger than 16.

So Arduino-LoRa uses "long" so after debugging, I guess it would trim down to 16.
[image]

So my suggested fix is to change this line
https://github.com/SMotlaq/LoRa/blob/0bf5cb70da10118b0dd6729fc7ed5118bd6b9e5e/LoRa/LoRa.c#L180

To

`	LoRa_setLowDaraRateOptimization(_LoRa, (long)((1 << _LoRa->spredingFactor) / ((double)BW[_LoRa->bandWidth])) > 16.0);`

My STM32L0 with this library is able to receive SF7-12 packets from Arduino-LoRa now.
[image]

SMotlaq, 2024-01-27:
Hi again,

Thanks for your suggestion to fix the issue. I agree that changing the line to use `long` and `>` operator would fix the issue.

Would you be willing to make this change in your forked repository and send a pull request? This way, we can keep the changes in your repository and I can merge them into the main repository.

If you're not comfortable making the changes yourself, I can certainly do it myself. However, I'd like to give you the opportunity to make the changes and get credit for your contribution.

Let me know either way, and we can move forward with the fix.

Thanks again!

IoTThinks, 2024-01-27:
Ok, let me creat a pull request tomorrow.

IoTThinks, 2024-01-28:
I created a Pull Request.
To close this issue.
Thanks a lot for your effort.

closed 2024-01-28 by IoTThinks